robotpy/robotpy-installer #8: Installer not working on newly imaged RoboRIO.

czhao39, 2017-02-02:
Here is the output when I run `python -m robotpy_installer install-robotpy` in Windows 10 when connected to a newly imaged RoboRIO:

> 20:31:22:613 INFO    : robotpy.installer   : RobotPy Installer 2017.1.2
20:31:22:613 INFO    : robotpy.installer   : -> caching files at C:\Users\czhao\OneDrive\Documents\frc2017-1123
20:31:24:161 INFO    : robotpy.installer   : Copying over the opkg cache...
20:31:24:172 INFO    : robotpy.installer   : Looking up hostname 'roborio-1123-frc.local'...
20:31:25:371 INFO    : robotpy.installer   : Connecting to robot via SSH at roborio-1123-frc.local
Remote working directory is /home/admin
local:C:\Users\czhao\OneDrive\Documents\frc2017-1123\opkg_cache\install_opkg.sh => remote:/home/admin/opkg_cache/install_opkg.sh
Using username "admin".
NI Linux Real-Time (run mode)
>
>Log in with your NI-Auth credentials.
>
>20:31:31:316 INFO    : robotpy.installer   : Copying over the pip cache...
20:31:35:291 INFO    : robotpy.installer   : Running installation...
20:31:35:291 INFO    : robotpy.installer   : Done.
netconsole-host - 1.0
netconsole-host already installed, continuing...
python36 - 3.6.0-r1
python36 already installed, continuing...
Collecting wpilib
  Could not find a version that satisfies the requirement wpilib (from versions: )
No matching distribution found for wpilib
20:31:42:236 ERROR   : robotpy.installer   : Command '['C:\\Users\\czhao\\OneDrive\\Documents\\frc2017-1123\\venv\\lib\\site-packages\\robotpy_installer\\win32\\plink.exe', '-pw', '', 'admin@roborio-1123-frc.local', 'bash opkg_cache/install_opkg.sh && /usr/local/bin/pip3 install --no-index --find-links=pip_cache --pre --upgrade wpilib robotpy-hal-base robotpy-hal-roborio robotpy-wpilib-utilities']' returned non-zero exit status 1

I get the same error when I run `python -m robotpy_installer install-pip wpilib`.

virtuald, 2017-02-02:
Did you execute 'download-robotpy' first? It would need to be executed from the same directory that this occurred in.

czhao39, 2017-02-02:
Yes. After imaging the RoboRIO, I ran `download-robotpy` and then `install-robotpy` in the same directory.

virtuald, 2017-02-02:
Hm. Is there anything in `C:\Users\czhao\OneDrive\Documents\frc2017-1123\pip_cache`?

czhao39, 2017-02-02:
Yep. Here's everything in `C:\Users\czhao\OneDrive\Documents\frc2017-1123\pip_cache`:

>pynetworktables-2017.0.4-cp35-none-any.whl
robotpy_hal_base-2017.0.6-cp35-none-any.whl
robotpy_wpilib_utilities-2017.0.3-cp35-none-any.whl
robotpy-hal-roborio-2017.0.6.tar.gz
wpilib-2017.0.6-cp35-none-any.whl

virtuald, 2017-02-02:
Oh, it's because it's downloading wheels. Closing as this is a duplicate of #3... will try to fix it soon.

For now, do a `py -3 -m pip install pip==8.1.2` and that should address the issue for now...

closed 2017-02-02 by virtuald