robotpy/robotpy-installer #15: Uses pip.main(), removed in pip 10.0.0

ErikBoesen, 2018-04-25:
I get this exception whenever I try downloading robotpy while using `pip` versions >=10.0.0. The only way to solve it that I've found (though, admittedly, given that I'm at a competition I've had scant time to try other solutions) is uninstalling `pip` and reinstalling version 9.x with `easy_install`.

```$ robotpy-installer download-robotpy
18:06:17:470 INFO    : robotpy.installer   : RobotPy Installer 2018.0.2
18:06:17:471 INFO    : robotpy.installer   : -> caching files at /private/tmp
Downloading https://www.tortall.net/~robotpy/feeds/2018/Packages
100%
Downloading http://download.ni.com/ni-linux-rt/feeds/2017/arm/ipk/cortexa9-vfpv3/Packages
100%
Traceback (most recent call last):
  File "/usr/local/bin/robotpy-installer", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/site-packages/robotpy_installer/installer.py", line 1129, in main
    retval = options.cmdobj(options)
  File "/usr/local/lib/python3.6/site-packages/robotpy_installer/installer.py", line 825, in download_robotpy
    return self.download_pip(self._create_rpy_pip_options(options))
  File "/usr/local/lib/python3.6/site-packages/robotpy_installer/installer.py", line 1022, in download_pip
    return pip.main(pip_args)
AttributeError: module 'pip' has no attribute 'main'```

auscompgeek, 2018-04-26:
Well crap.

I didn't realise that was considered an internal interface.

auscompgeek, 2018-04-26:
Realistically, I think we should do a subprocess call instead of calling pip's main (which assumes it has control of the entire interpreter).

Meanwhile, I think the best advice we can give to teams going to Detroit is "don't upgrade to pip 10".

virtuald, 2018-04-26:
Yeah, it seems that's what they recommend on their docs.

virtuald, 2018-04-26:
Also, for future reference, you could have used pip to install an older version of pip

```
# Windows
py -3 -m pip install 'pip < 10'

# OSX/Linux
pip3 install 'pip < 10'
```

closed 2018-04-26 by auscompgeek

ErikBoesen, 2018-04-26:
Yeah, I know, for some reason that wasn't working and I had to uninstall completely for the version to change. Probably just an issue with my setup.

ErikBoesen, 2018-04-26:
Thanks for the quick response.

commit f11cb86c (2018-04-26): Fix downloading with pip 10

- Fixes #3
- Fixes #15