robotpy/robotpy-installer #34: What if we distributed the installer as a standalone exe?

virtuald, 2019-12-23:
... instead of distributing a packaged RobotPy zipfile? The install docs would include that as part of your introduction to RobotPy. Could use pyinstaller for this.

One advantage of this approach is that then teams wouldn't need to install python on their non-development computers.

Probably would need a GUI version at some point.

virtuald, 2019-12-25:
Because paramiko (https://github.com/robotpy/robotpy-installer/issues/31), this would make sense. 

Also, it probably makes sense to no longer distribute a release package, but have people use the installed version of robotpy-installer?

auscompgeek, 2019-12-28:
pyinstaller is super-slow though. Would PyOxidizer or other tools work for our use-case?

Suave101, 2023-01-08:
PyInstaller has many options that are difficult to navigate, you could use [auto-py-to-exe](https://pypi.org/project/auto-py-to-exe/) to make the development of a standalone exe faster.

virtuald, 2023-01-09:
Now that we have robotpy-meta, I don't feel there's as strong a need for this anymore.

closed 2023-01-09 by virtuald