robotpy/robotpy-installer #42: robotpy installer not working

cody-hanks, 2020-02-11:
New linux 18.04 LTS 
New install of pyfrc after some work 
New image of roborio   

Command works
_python3 robotpy_installer download-robotpy_  

Command fails 
_python3 robotpy_installer install-robotpy_

12:29:04:414 INFO    : robotpy.installer   : Finding robot for team 1432
12:29:05:499 INFO    : robotpy.installer   : -> Robot is at roboRIO-1432-FRC.local
12:29:05:499 INFO    : robotpy.installer   : Connecting to robot via SSH at roboRIO-1432-FRC.local
12:29:07:639 INFO    : paramiko.transport  : Connected (version 2.0, client OpenSSH_7.6)
12:29:07:844 INFO    : paramiko.transport  : Auth banner: b'NI Linux Real-Time (run mode)\n\nLog in with your NI-Auth credentials.\n\n'
12:29:07:845 INFO    : paramiko.transport  : Authentication (password) successful!
IMAGEVERSION = "FRC_roboRIO_2020_v10"
Downloading http://localhost:33215/opkg_cache/python38_3.8.1-r0_cortexa9-vfpv3.ipk.
**12:29:08:665 ERROR   : robotpy.installer   : Command** (IV=$(grep IMAGEVERSION /etc/natinst/share/scs_imagemetadata.ini); echo $IV; [ "$IV" == 'IMAGEVERSION = "FRC_roboRIO_2020_v10"' ] || (echo '-> ERROR: installer requires RoboRIO image 2020_v10! Use --ignore-image-version to force install' && /bin/false)) && echo 'set -e
PACKAGES=()
DO_INSTALL=0
if ! opkg list-installed | grep -F "python38 - 3.8.1-r0"; then
    PACKAGES+=("http://localhost:33215/opkg_cache/python38_3.8.1-r0_cortexa9-vfpv3.ipk")
    DO_INSTALL=1
else
    echo "python38 already installed"
fi
if ! opkg list-installed | grep -F "python38-robotpy-wpiutil - 2020.2.2.0"; then
    PACKAGES+=("http://localhost:33215/opkg_cache/python38-robotpy-wpiutil_2020.2.2.0_cortexa9-vfpv3.ipk")
    DO_INSTALL=1
else
    echo "python38-robotpy-wpiutil already installed"
fi
if ! opkg list-installed | grep -F "python38-pyntcore - 2020.2.2.2"; then
    PACKAGES+=("http://localhost:33215/opkg_cache/python38-pyntcore_2020.2.2.2_cortexa9-vfpv3.ipk")
    DO_INSTALL=1
else
    echo "python38-pyntcore already installed"
fi
if ! opkg list-installed | grep -F "python38-robotpy-hal - 2020.2.2.0"; then
    PACKAGES+=("http://localhost:33215/opkg_cache/python38-robotpy-hal_2020.2.2.0_cortexa9-vfpv3.ipk")
    DO_INSTALL=1
else
    echo "python38-robotpy-hal already installed"
fi
if ! opkg list-installed | grep -F "python38-wpilib - 2020.2.2.7"; then
    PACKAGES+=("http://localhost:33215/opkg_cache/python38-wpilib_2020.2.2.7_cortexa9-vfpv3.ipk")
    DO_INSTALL=1
else
    echo "python38-wpilib already installed"
fi
if [ "${DO_INSTALL}" == "0" ]; then
    echo "No packages to install."
else
    opkg install  ${PACKAGES[@]}
fi' > install_opkg.sh && bash install_opkg.sh && rm install_opkg.sh && ([ -x /usr/local/bin/pip3 ] || exit 87) && /usr/local/bin/pip3 install --no-index --find-links=http://localhost:33215/pip_cache/ --pre --upgrade --no-deps pynetworktables robotpy-wpilib-utilities **returned non-zero error status 255**

virtuald, 2020-02-11:
There's probably something useful in stderr. Please upgrade to robotpy-installer 2020.0.3 and see if the error messages are more informative.

cody-hanks, 2020-02-11:
That would be 
pip3 install robotpy-installer=2020.0.3 

just making sure i have the command correct I am working on ubuntu linux

virtuald, 2020-02-11:
`pip3 install robotpy-installer==2020.0.3`

virtuald, 2020-02-12:
Did you figure out your issue?

closed 2020-03-24 by virtuald