robotpy/robotpy-installer #55: Package source 2021 failed

yuhao1118, 2021-08-12:
Hi, when I follow the [instruction](url) to install package on RoboRIO, I got the error below:
```
Downloading https://www.tortall.net/~robotpy/feeds/2021/Packages
Traceback (most recent call last):
  File "/Users/liyuhao/miniforge3/bin/robotpy-installer", line 8, in <module>
    sys.exit(main())
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/robotpy_installer/installer.py", line 489, in download_python
    ctx.forward(opkg_download, packages=[_ROBOTPY_PYTHON_VERSION])
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/core.py", line 784, in forward
    return __self.invoke(__cmd, *args, **kwargs)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/click/decorators.py", line 38, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/robotpy_installer/installer.py", line 289, in opkg_download
    opkg.update_packages()
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/robotpy_installer/opkgrepo.py", line 48, in update_packages
    _urlretrieve(pkgurl, feed["db_fname"], True)
  File "/Users/liyuhao/miniforge3/lib/python3.9/site-packages/robotpy_installer/utils.py", line 70, in _urlretrieve
    with contextlib.closing(urllib.request.urlopen(req)) as rfp:
  File "/Users/liyuhao/miniforge3/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/Users/liyuhao/miniforge3/lib/python3.9/urllib/request.py", line 523, in open
    response = meth(req, response)
  File "/Users/liyuhao/miniforge3/lib/python3.9/urllib/request.py", line 632, in http_response
    response = self.parent.error(
  File "/Users/liyuhao/miniforge3/lib/python3.9/urllib/request.py", line 561, in error
    return self._call_chain(*args)
  File "/Users/liyuhao/miniforge3/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/Users/liyuhao/miniforge3/lib/python3.9/urllib/request.py", line 641, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
```

It turns out that https://www.tortall.net/~robotpy/feeds/2021/Packages is 404.

auscompgeek, 2021-08-12:
It looks like you have an old version of robotpy-installer. Could you please follow the instructions at https://robotpy.readthedocs.io/en/stable/install/computer.html to upgrade your RobotPy installation?

auscompgeek, 2021-08-12:
Oh wait, that's still used to install Python, right. Hmm, @PeterJohnson did something happen to our files?

yuhao1118, 2021-08-13:
> Oh wait, that's still used to install Python, right. Hmm, @PeterJohnson did something happen to our files?

Hi, it looks like the 2020 source feed still works: https://www.tortall.net/~robotpy/feeds/2020/Packages

virtuald, 2021-08-13:
The content needs to be resynced, should hopefully be available soon (or at least, by tomorrow).

PeterJohnson, 2021-08-13:
Should be fixed now.

closed 2021-08-13 by virtuald