robotpy/robotpy-installer #57: Detect SSL errors and suggest certifi upgrade

virtuald, 2022-01-09:
Reference: https://www.chiefdelphi.com/t/robotpy-module-not-found/399156/2

I'm guessing that's their problem, as I believe pip uses certifi for certificates instead of the system store.

virtuald, 2022-01-10:
Well, maybe not. Another thing we could do is add an option to use certifi, it seems that python gives you the option to do that.

virtuald, 2022-01-10:
Turns out if you browse to the url using MS edge the error goes away. Maybe we just tell the user to do this? We already have a special check of this sort for OSX.

TheTripleV, 2022-01-10:
Should it just automatically try certifi if the original version fails on that specific error?

virtuald, 2022-01-17:
That seems reasonable.

virtuald, 2023-11-13:
I think we do this now.

closed 2023-11-13 by virtuald