robotpy/robotpy-installer #64: [BUG]: deploy fails with socket.gaierror

virtuald, 2022-03-21:
### Problem description

An error occurs that has `socket.gaierror: [Errno 8] getaddrinfo failed` in it.

I ran into this today on my team's laptop on Windows, but I've never ran into it on OSX/Linux. It kept selecting the *.local address, but didn't seem to have actually resolved it so it passes the DNS name to paramiko for the connection. When paramiko tries to use it, the connection fails for some reason.

If you look carefully at the `_resolve_addr` function, if none of the addresses are a SOCK_STREAM address then it will just return the original name. I'm not really sure why this is the case.

I don't have a Windows machine at home so I don't think I'll be able to diagnose this anytime soon. It would be great if someone else who has this issue could give the whole console output when running `py robot.py deploy --debug`.

### Operating System

Windows

peter9477, 2022-03-21:
If I'm reading the `RobotFinder` code correctly, it also appears to be waiting for all threads to terminate before finishing, even after the first successful result is found.  And for the "resolve=False" addresses, it attempts an SSH connection with a 10s timeout.  Both those things seem somewhat undesirable.

peter9477, 2022-03-21:
Oh, sorry, correct. I misread it as though it were an "or" condition.

PR #66 (2022-12-09): More efficient finding
- Fixes #64
- Maybe https://github.com/robotpy/pyfrc/issues/204 too? 

@TheTripleV any chance you can try this out on Windows?
closed 2022-12-09 by virtuald