robotpy/robotpy-installer #70: Add package synchronization mechanism for local/remote python environment

virtuald, 2023-05-25:
Idea is something like this:

* User specifies their dependencies in pyproject.toml
* When pyfrc deploys, it ensures your remote environment matches your environment in pyproject.toml
* robotpy-installer looks in your pyproject.toml to know what version to download to your offline cache, and optionally to your local installation too
  * call the command 'sync'? 

Vendors are TBD. Probably makes sense to keep vendors in vendordep jsons and integrate that with this somehow... but maybe not.

Python vscode extension would need to know how to update pyproject.toml to upgrade to a specific wpilib wheel. Might just make sense to have that be an installer command?

virtuald, 2023-10-12:
Here's another version of the same idea:

> well ok, let's go with it for a second -- so let's say you force users to write a pyproject.toml
then you could have some robotpy-specific backend that as part of the build process builds a wheel -- and if there's something in requirements that isn't present in the local cache, it grabs it from the internet 
then the deploy thing takes that wheel and sends it to the rio, which automatically installs whatever the requirements are
the deploy thing could probably just build the wheel and deploy it in a single step, it probably wouldn't take very long?
it would just suck if someone updated the requirements and forgot to run a build
but I guess that you'd have that issue anyways?
it's sorta nice too because you could take the wheel and archive it or something to know what you deployed
I think the issue you'll likely run into is someone is going to want to use poetry or whatever the new hotness is
so you'd either have to make it co-exist with that or force users to not do that
I dunno, I think if it were done well it would be pretty neat, and solve some of the issues that we have around requirements
there are probably downsides too

virtuald, 2023-12-21:
Thinking through the workflow a user will take from a clean python installation:

* install minimal dependencies: `pip install robotpy`

Then they make a robot.py, add the `if __name__ == __main__` thing + `wpilib.run`. Add a `pyproject.toml` with the following:

```toml
[tool.robotpy]

# equivalent to `robotpy==2024.0.0b4`
robotpy_version = "2024.0.0b4"

# equivalent to `robotpy[cscore, ...]`
robotpy_extras = ["cscore"]

# Other pip installable requirement lines
requires = [
  "numpy"
]

```

Ideally, you'd like them to do:

```
python robot.py sync [--user]
```

And that would set up their local environment with the correct pip packages, and *also* download python and all of the packages for roborio.

I think you'd also have to provide a robotpy-installer command for this:

```
robotpy-installer sync [--user]
```

User does normal development, and then they do:

```
python robot.py deploy
```

Before doing the deploy...

* It will check to see if python installed on the rio, install it if not
* It will check to see if the local pip packages match whatever is in `pyproject.toml` and complain if not
  * `pip list --format json` for inspecting local environment
* It will check to see if the pip packages on the rio match whatever is in `pyproject.toml`, and if not then it will try to install them (should it delete packages not in `pyproject.toml`?).
  * Probably should try to detect if the packages are in the local cache?
  * `pip list --format json` for inspecting remote packages

Then it will do the normal deploy stuff.

---

We would also need to support users that don't have a `pyproject.toml`.

If `pyproject.toml` didn't exist, I think we would just write one with the current version of robotpy installed in their local environment, then do whatever action that required the `pyproject.toml`.

---

The synchronization behavior above is likely covered by `pip-tools`, but it seems complicated and has a lot of dependencies, so I don't think it's a good candidate for running on the RoboRIO.

TheTripleV, 2023-12-24:
I think it should make a virtualenv and generate a vscode python config. That way packages are project specific like cpp and java

virtuald, 2024-01-01:
> I think it should make a virtualenv and generate a vscode python config. That way packages are project specific like cpp and java

I would not want to do that by default. I think this could be a separate action `setup-vscode-venv` or something similar?

virtuald, 2024-01-06:
Fixed in https://github.com/robotpy/robotpy-installer/pull/79

commit 0e0db4c9 (2024-01-06): Install project requirements at deploy time

- Fixes #70
commit 59cc6959 (2024-01-06): Install project requirements at deploy time

- Fixes #70
commit 312af5ab (2024-01-06): Install project requirements at deploy time

- Fixes #70
closed 2024-01-06 by virtuald