robotpy/robotpy-installer #107: robotpy init should set up a basic .gitignore

virtuald, 2024-01-20:


auscompgeek, 2024-01-21:
Including some of the WPILib template gitignore would be a good idea: https://github.com/wpilibsuite/vscode-wpilib/blob/main/vscode-wpilib/resources/gradle/shared/.gitignore

sciencewhiz, 2024-10-29:
I think this is a better solution then wpilibsuite/vscode-wpilib#710

auscompgeek, 2025-01-03:
For the benefit of any contributors who'd like to pick this up: we should include https://github.com/github/gitignore/blob/main/Python.gitignore as well as the WPILib template's.

Needs to be added into https://github.com/robotpy/robotpy-installer/blob/ed6870c0242fe4fca81469684f0792ad5ce9629a/robotpy_installer/cli_init.py#L22

If a new non-Python file is added, it'll need to be added to `MANIFEST.in` to be included in the package. https://github.com/robotpy/robotpy-installer/blob/ed6870c0242fe4fca81469684f0792ad5ce9629a/MANIFEST.in#L1

closed 2025-02-12 by virtuald