robotpy/robotpy-installer #118: Remove robotpy_extras from pyproject.toml

virtuald, 2024-10-16:
This is a bit more work than it sounds, need to provide a good alternative mechanism that's easy to update.

gerth2, 2024-12-17:
No major complaint from a user-end, it's trivial to move it over in the toml file. It was always slightly confusing why there were two options.

Inject a comment by default that points at a doc page to how to update the field with common vendor dependencies might be a quick solution to the tool question. I'm not fully visualizing what value the tool has over just manually editing the file, and seeing the errors in robotpy if conflicts are added.

TheTripleV, 2024-12-17:
I'm still a proponent of having a list to install "all". I like having a oneshot command to get everything under the sun

virtuald, 2025-01-04:
I want to get rid of this, but pushing it to 2026.

virtuald, 2025-01-04:
it does occur to me that there is something that robotpy-extras is useful for (but we should rename it to optional_components or something more obvious like that) -- namely, all of the optional wpilib pieces that we don't install by default (cscore, apriltag, etc). That's because they're all robotpy and are correlated together

I think all the vendor stuff should be removed from it and figure out a way to make 'requires' more useful.

PR #151 (2025-10-08): Replace 'robotpy_extras' with 'components'
- This is mostly cosmetic, but hopefully communicates the purpose better
- Fixes #118
closed 2025-10-08 by virtuald