robotpy/robotpy-installer #127: Add single-command mechanism

virtuald, 2024-12-26:
Should add something to allow users to execute a single arbitrary command, so that when we need them to do something manual we can just give them a command instead of "figure out how to ssh to your machine and do this".

Suave101, 2025-01-05:
Maybe we could implement a command called something like `ssh` or `manual` using a socket connection and supports arguments for the [lvuser account](https://docs.wpilib.org/en/stable/docs/software/roborio-info/roborio-ssh.html#lvuser) and for the [admin account](https://docs.wpilib.org/en/stable/docs/software/roborio-info/roborio-ssh.html#admin). From there a SSH console opens up so that the user can make changes on the roborio without Putty.

Should I open a Pull Request working on the following suggestions?

virtuald, 2025-01-05:
Implementing a fully interactive ssh session correctly is a lot more work than it sounds and I suspect would involve a lot of maintenance over time. If a user needs a fully interactive ssh session then why wouldn't they just use an SSH client?

Generally, users shouldn't need to SSH into the robot directly when using RobotPy, and when they need that then it's a good opportunity for them to learn how to do that.

The thought that I had with this issue is that very often we'll tell users "run such a command on the rio to fix something or tell us something about the rio", and in those limited situations it's really inconvenient to tell a user how to setup an SSH client. Implementing something that can execute a single command is extremely simple and would make those sorts of user/support interactions much easier to manage.

If you want to implement the single command thing, happy to review it.

auscompgeek, 2025-01-05:
> without Putty

Windows ships with the OpenSSH client, so nobody needs PuTTY to SSH anyway.

PR #131 (2025-02-04): Add simple SSH command for diagnostic purposes
- Fixes #127
commit da21470f (2025-02-04): Add simple SSH command for diagnostic purposes

- Fixes #127
closed 2025-02-04 by virtuald